ploopyco/headphones #2: Documentation: Material sourcing improvements

qwertymodo, 2023-02-10:
[Appendix D](https://github.com/ploopyco/headphones/wiki/Appendix-D%3A-Hardware) could use some additional details to properly source materials.

First up, the stockinette only lists a single size, 1.5", but your kits for sale include 2 different sizes, 2" and 3".  So, there's a mismatch there.

Some additional details on the foam would be really helpful, e.g. what kind of material, the density/pressure you use (gives better context to compare against, rather than just "get the lowest possible compression pressure you can find")

Sourcing information for the magnets would be helpful.  I can't seem to find the correct size anywhere.  Even the specialty magnet distributors I've looked at don't seem to carry that exact size.

The rest of the materials seem easy enough to source from the provided information.

megamaniac, 2023-02-10:
I would definitely like to see this as well - for example a 25x5x5 magnet is much easier for me to get hold of. Will this still work in the same way or is 30mm an absolute requirement?

If the dimensional values of the parts can be modified it would be amazing to note this in the descriptions for ease of sourcing.

ploopyco, 2023-02-17:
@qwertymodo I fixed the documentation to include more info about the stockinettes, as well as the foams.

I used Aliexpress to find different magnet sizes and found three vendors supplying magnets in the correct size. Is Aliexpress an option where you are?

@megamaniac The magnet size is not an absolute requirement, of course. The deflection of the PCB is going to change, though, if magnets of different sizes are used.

qwertymodo, 2023-02-17:
AliExpress is definitely an option, and one that I searched before posting here. Unfortunately with things like this it can sometimes be difficult to find exactly what you're looking for, buried in pages of similar items that just slightly aren't correct. Do you happen to have a link to one of those magnet vendors you found?

closed 2023-07-28 by ploopyco