ploopyco/headphones #3: Option to Purchase Just the PCBs

Wgelyjr, 2023-02-10:
I'd love to manufacture the rest of the parts - it'd be a fun challenge. 

I would not like to wrangle a manufacturer into a production run, 

I would like to support this awesome project. Mark em up, I don't care, I'll still buy them.

WhosManzIsThis, 2023-02-10:
Agreed,  it would be nice to have an option in between a DIY full kit and sourcing all the parts ourselves.  For example, I happen to have access to 3D printers, but not too many options when it comes to pcb manufacturing, thus having a pre-assembled circuit boards stand-alone purchase option would be ideal.  I really hope the individuals at this company see this thread.

qwertymodo, 2023-02-12:
Bare PCB's only and a "print your own" kit would both be really nice purchase options.  As somebody who has a printer and a lot of really cool filament options, I'd definitely be up for buying a full kit if it didn't feel so wasteful to buy and ship printed parts that I would just throw out.  I was originally thinking of doing a fully self-sourced build, but some of the materials are a bit tricky to find, so a print your own kit would definitely be a good option for people like me.

ploopyco, 2023-02-17:
For the moment, I'm going to continue supplying kits with the 3D-printed parts, only. I do have multiple reasons for sticking with this decision, [which you can read about here](https://blog.ploopy.co/why-we-dont-offer-kits-without-3d-printed-parts-52).

closed 2023-02-17 by ploopyco

Wgelyjr, 2023-02-18:
I see your reasons, I'm not sure they justify the decision, but it's yours to make. I hope you reconsider at some point, but I bought your kit anyways, and I look forward to receiving it. I appreciate your hard work on this project.