ploopyco/headphones #4: Gerber files

qwertymodo, 2023-02-11:
As far as I can find, Altium doesn't offer any free means of converting their design files into gerbers, would it be possible to include them in the repo?

sirus20x6, 2023-02-14:
to convert altium files into gerbers you have to create an output generator. Then under that output generator file you go to fabrication outputs and you'll want gerber and NC drill files.

qwertymodo, 2023-02-14:
> to convert altium files into gerbers you have to create an output generator. Then under that output generator file you go to fabrication outputs and you'll want gerber and NC drill files.

Yes, but first you have to spend several hundred dollars to purchase Altium. Locking the files behind a proprietary format that has no free way of accessing them isn't really in the spirit of open source.

sirus20x6, 2023-02-14:
I can do it from work tomorrow. I'll also export them as ascii files which
I think can be imported to a few different editors.

sirus20x6, 2023-02-15:
[ploopyco.zip](https://github.com/ploopyco/headphones/files/10746818/ploopyco.zip)
Here are the exported gerbers, nc drill files, i also created and included the outjob file and a draftsman file. You'll find exports for ascii versions of the schematics and pcb as well as specctra design, sisoft, p-cad, orcad, ansoft, hyperlynx files.
I've also generated cad files: step, idf, dwg

ploopyco, 2023-02-17:
I believe that KiCAD has an importer for Altium tools.

qwertymodo, 2023-02-17:
I tried a few tools that claimed to support Altium import and they rejected the files because they were not in an ASCII format.

sirus20x6, 2023-02-17:
> I tried a few tools that claimed to support Altium import and they rejected the files because they were not in an ASCII format.

use the zip in my above comment. I have ascii files in there.

closed 2023-07-28 by ploopyco