ploopyco/headphones #5: unmerge 3d files

sirus20x6, 2023-02-14:
It would really be helpful if every 3d part was in its own file. I'm not sure what we're supposed to do with them all merged together, but I can't give this to my 3d printer

sirus20x6, 2023-02-16:
Been poking at this for a while now. even in the step file everything is fused into one object. you can't actually 3d print it this way.

ploopyco, 2023-02-17:
Cura and PrusaSlicer are two free tools that have both been tested and verified as having the ability to "split" the file into the individual pieces.

As for the STEP file, FreeCAD and Blender are also two free tools that have been tested and verified as having the ability to split the file into the individual components for modification.

closed 2023-02-17 by ploopyco

sirus20x6, 2023-02-17:
I was unable to split them in either freecad or blender

sirus20x6, 2023-02-17:
I just tried cura. it is unable to split the stl because all the geometry is together under one object. cura is unable to load the stp file

sirus20x6, 2023-02-17:
I just tried prusa slicer as well. it was able to load the stp file, but all the geometry again was under one single object which means it cant be split. it has to be printed as is. in order to print any of these they have to each be their own object which can be done with stp, but is not in this step file.