ploopyco/headphones #7: DAC always resampling to 48kHz 16bit

thejevans, 2023-03-14:
I just built my ploopy headphones today. Upon testing, it seems that the DAC always resamples audio to 48kHz 16bit. I made sure to update to the pm5 firmware during testing and the problem still exists. I've tested with sample rates above and below 48kHz and the result is always the same. I tested on multiple linux machines with pipewire and with USB Audio Player Pro on Android.

thejevans, 2023-03-14:
I just noticed the sentence on the product page saying that you all decided to force resampling. Are there downsides to allowing sample rate switching?

ploopyco, 2023-03-23:
The 48kHz/16-bit rate is set by the USB device descriptor [here](https://github.com/ploopyco/headphones/blob/cca0639485fa7383ba7714f32e0514ca87521c97/firmware/code/run.c#L258). So yes, it always downsamples higher bit-rate audio, but that happens before it even gets to the DAC.

The only downside to doing a higher bit rate is that the EQ will have fewer bands. Since all of the EQ is done in the digital domain, a higher bit rate obviously means more bits, and the EQ is primarily limited by processing power. So, more audio quality means less EQ. That's about the only downside (though it's a major one).