ploopyco/headphones #9: Updated firmware binary? 🙏

13tales, 2023-04-06:
Minor issue: could we get an updated compiled firmware?

I'm keen to try out @george-norton's fix to the firmware to see if it has any bearing at all on the USB audio issues affecting recent Macs. 

I've tried building it myself, but I'm thinking there may be an issue with my development environment. The resulting noise from 30cm away demonstrated the timeless pedagogical wisdom of the Ploopy firmware-flashing rhyme.

13tales, 2023-04-06:
*Update:* There are definitely problems with my development environment; I've tried building and flashing an earlier commit of the firmware code, with the same result (ie. no sound besides occasional loud screeching.) Would appreciate any insights folks might have on what might be wrong and how to rectify it.

I have some initial issues convincing CMake to find the correct compiler (`arm-none-eabi-gcc`, as specified by the Raspberry Pi Pico PDF), but once I solve those by editing the CMake cache, everything *appears* to work fine; no errors amongst the build output, and I end up with a `ploopy_headphones.uf2` file in my `/build` folder. Flashing it only leads to unpleasant noises and sorrow, however 😬

One discrepancy I've noticed that seems like it might be relevant is that *my* firmware build seems suspiciously larger than either the current binary in `master` or the alpha build you emailed recently. Both of those were ~90kb, whereas mine is coming out at 161kb.

This is the state of VSCode/CMake when I trigger the build (mode: `[Release]`, compiler: `[GCC 10.3.1 arm-none-eabi]`, target: `[all]`)
[image]

Here's the build output from VS Code:
```
[main] Building folder: pico 
[build] Starting build
[proc] Executing command: /opt/homebrew/bin/cmake --build /Users/tom.armstrong/Workspace/pico/build --config Release --target all -j 10 --
[build] [  2%] Creating directories for 'PioasmBuild'
[build] [  2%] Creating directories for 'ELF2UF2Build'
[build] [  3%] Building ASM object pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[build] [  5%] No download step for 'ELF2UF2Build'
[build] [  6%] No download step for 'PioasmBuild'
[build] [  7%] No update step for 'ELF2UF2Build'
[build] [ 10%] No update step for 'PioasmBuild'
[build] [ 10%] No patch step for 'ELF2UF2Build'
[build] [ 11%] Linking ASM executable bs2_default.elf
[build] [ 12%] Performing configure step for 'ELF2UF2Build'
[build] [ 13%] No patch step for 'PioasmBuild'
[build] [ 15%] Performing configure step for 'PioasmBuild'
[build] -- Configuring done (0.0s)
[build] loading initial cache file /Users/tom.armstrong/Workspace/pico/build/pico-sdk/src/rp2_common/pico_cyw43_driver/pioasm/tmp/PioasmBuild-cache-Release.cmake
[build] -- Generating done (0.0s)
[build] -- Build files have been written to: /Users/tom.armstrong/Workspace/pico/build/elf2uf2
[build] -- Configuring done (0.0s)
[build] [ 16%] Performing build step for 'ELF2UF2Build'
[build] -- Generating done (0.0s)
[build] -- Build files have been written to: /Users/tom.armstrong/Workspace/pico/build/pioasm
[build] [ 17%] Performing build step for 'PioasmBuild'
[build] [ 17%] Built target bs2_default
[build] [ 18%] Generating bs2_default.bin
[build] [100%] Built target elf2uf2
[build] [ 20%] No install step for 'ELF2UF2Build'
[build] [ 21%] Generating bs2_default_padded_checksummed.S
[build] [ 22%] Completed 'ELF2UF2Build'
[build] [100%] Built target pioasm
[build] [ 22%] Built target ELF2UF2Build
[build] [ 24%] No install step for 'PioasmBuild'
[build] [ 24%] Built target bs2_default_padded_checksummed_asm
[build] [ 25%] Completed 'PioasmBuild'
[build] [ 25%] Built target PioasmBuild
[build] [ 26%] Generating i2s.pio.h
[build] [ 26%] Built target ploopy_headphones_i2s_pio_h
[build] [ 27%] Building C object CMakeFiles/ploopy_headphones.dir/run.c.obj
[build] [ 29%] Building C object CMakeFiles/ploopy_headphones.dir/ringbuf.c.obj
[build] [ 31%] Building C object CMakeFiles/ploopy_headphones.dir/user.c.obj
[build] [ 32%] Building C object CMakeFiles/ploopy_headphones.dir/Users/tom.armstrong/Workspace/pico/pico-sdk/src/rp2_common/hardware_pio/pio.c.obj
[build] [ 34%] Building C object CMakeFiles/ploopy_headphones.dir/fix16.c.obj
[build] [ 34%] Building C object CMakeFiles/ploopy_headphones.dir/i2s.c.obj
[build] [ 35%] Building C object CMakeFiles/ploopy_headphones.dir/Users/tom.armstrong/Workspace/pico/pico-sdk/src/rp2_common/pico_platform/platform.c.obj
[build] [ 36%] Building C object CMakeFiles/ploopy_headphones.dir/bqf.c.obj
[build] [ 37%] Building C object CMakeFiles/ploopy_headphones.dir/Users/tom.armstrong/Workspace/pico/pico-sdk/src/rp2_common/hardware_claim/claim.c.obj
[build] [ 39%] Building C object CMakeFiles/ploopy_headphones.dir/Users/tom.armstrong/Workspace/pico/pico-sdk/src/rp2_common/hardware_sync/sync.c.obj
[build] [ 40%] Building C object CMakeFiles/ploopy_headphones.dir/Users/tom.armstrong/Workspace/pico/pico-sdk/src/rp2_common/hardware_gpio/gpio.c.obj
[build] [ 41%] Building C object CMakeFiles/ploopy_headphones.dir/Users/tom.armstrong/Workspace/pico/pico-sdk/src/rp2_common/hardware_irq/irq.c.obj
[build] [ 43%] Building ASM object CMakeFiles/ploopy_headphones.dir/Users/tom.armstrong/Workspace/pico/pico-sdk/src/rp2_common/hardware_irq/irq_handler_chain.S.obj
[build] [ 44%] Building C object CMakeFiles/ploopy_headphones.dir/Users/tom.armstrong/Workspace/pico/pico-sdk/src/common/pico_sync/sem.c.obj
[build] [ 45%] Building C object CMakeFiles/ploopy_headphones.dir/Users/tom.armstrong/Workspace/pico/pico-sdk/src/common/pico_sync/lock_core.c.obj
[build] [ 46%] Building C object CMakeFiles/ploopy_headphones.dir/Users/tom.armstrong/Workspace/pico/pico-sdk/src/common/pico_sync/mutex.c.obj
[build] [ 48%] Building C object CMakeFiles/ploopy_headphones.dir/Users/tom.armstrong/Workspace/pico/pico-sdk/src/common/pico_sync/critical_section.c.obj
[... cut ...]
[build] [ 97%] Building C object CMakeFiles/ploopy_headphones.dir/Users/tom.armstrong/Workspace/pico/pico-extras/src/rp2_common/usb_device/usb_stream_helper.c.obj
[build] [ 98%] Building C object CMakeFiles/ploopy_headphones.dir/Users/tom.armstrong/Workspace/pico/pico-sdk/src/rp2_common/pico_fix/rp2040_usb_device_enumeration/rp2040_usb_device_enumeration.c.obj
[build] [100%] Linking CXX executable ploopy_headphones.elf
[build] [100%] Built target ploopy_headphones
[build] Build finished with exit code 0```

13tales, 2023-04-07:
> I have pushed a simple script here that builds everything inside a simple docker container, it should provide a consistent build environment: https://github.com/george-norton/headphones/tree/docker_build_script/firmware/docker

Thanks! Am I missing something; won't it also need to pull in the Pico SDK and Pico Extras?

13tales, 2023-04-07:
@george-norton Very nice! It worked, thanks.

I did find it was necessary to add `RUN git config --global --add safe.directory "*"` to the Dockerfile, or I received an error along the lines of `"fatal: detected dubious ownership in repository"` from Git when it was attempting to fetch the Pico SDK.

Still very curious about why my attempting to build with VS Code was apparently succeeding, but resulting in those broken oversized binaries 🤔

13tales, 2023-04-07:
It's too early to say conclusively whether the firmware change reduces the Mac M1-related USB audio issues. 
I've definitely noticed some occurrences of it since flashing the new firmware just now, but it seems like it *might* possibly have reduced the frequency and severity? 
Memory consumption seems to be a cause of the issue, and I'm not working today, so don't have quite my usual burden of docker containers, nodeJS processes, and browser tabs open and competing for resources.

It seems plausible that power consumption might be a factor, based on Colin's speculation, and my testing with the only other USB audio device that I have around (the dongle for a Logitech gaming headset, which *doesn't* suffer from the issue). Is this update likely to have reduced the power draw of the Ploopy DAC at all?

13tales, 2023-04-13:
I had a poke around in the MacOS audio settings; both devices accept 16bit 48khz. 
I think this is a low-level issue with MacOS and the M1 hardware in particular; the older and *less*-powerful Intel machines don't seem to suffer from it. [Here's one of many threads on the issue](https://developer.apple.com/forums/thread/677383?page=2), if you're interested.

Firmware programming and USB audio are well outside my realm of competence, so I'm not sure what to look for re: audio clocking. I've done my best to inspect both devices using some of the MacOS utilities and try and figure out what's different. The only thing I've spotted so far is that the `bMaxPacketSize0` property differs between them; it's `0x40` for the Ploopy DAC, `0x10` for the Logitech. No idea whether that could be a factor.

The underlying issue obviously rests with Apple, but I am curious if there's a workaround possible within the Ploopy firmware. 

I've tried meddling with the ring buffer size within the Ploopy firmware, doubling it to 64kb—I only use these to listen to music, so I don't really care about latency—but it doesn't fix the issue. It *might* have decreased its frequency, but without measurements to back that up, I think it's probably the placebo effect on my part.

george-norton, 2023-04-13:
@ploopyco The docker build script is a bit quick and dirty, but it works and it should ensure everyone produces identical binaries when they checkout a particular revision of the code. Potentially you could also push prebuilt images to some docker repository and speed up the build process.
Once you have had a play with it, let me know if you are still interested in a PR.

13tales, 2023-04-13:
> @13tales I'll update the repo with an updated firmware. I've been lazy with it, but I'll get it done.

@ploopyco I'm happily building away with the Docker script now, but much appreciated, thanks 🙏 Feel free to close this issue.

I doubt I'm the only person who's going to run into the irritating MacOS/M1 issue, and I'm holding out hope that a Mac user who knows what they're doing might be able to figure out a firmware-side workaround for it. Might create another issue thread as a home for discussion about it, if that's okay by you.

ploopyco, 2023-04-14:
> Might create another issue thread as a home for discussion about it, if that's okay by you.

Fine with me.