ploopyco/headphones #11: Headband flexbar symbols different in wiki

JRud52, 2023-04-18:
[This part of the wiki](https://github.com/ploopyco/headphones/wiki/Part-4%3A-Headband#-step-18-select-a-headband-clamping-strength) lists the headband flexbars as two types distinguished by:

> The Headband Flexbar with a solid circle has a higher clamping force. That means that it will feel tighter on your head.
> The Headband Flexbar with a hollow circle has a lower clamping force. That means that it will feel looser on your head.

I just went through the build process and noticed that the symbols on the flexbars that I received were different than stated in the wiki section:
The tighter flexbar was a hollow circle with a square
The looser flexbar was a hollow circle with a circle. 

It took me a while to figure out which flexbar was which, so it would be good to update that section with the right symbols.

JRud52, 2023-04-18:
The overall build process was easy to follow aside from that one section with the flexbars. They were a lot of fun to build and they sound great!

ploopyco, 2023-04-18:
Updated the wiki. Thanks for pointing this out!

closed 2023-04-18 by ploopyco