ploopyco/headphones #16: Documentation references unnumbered sections

mrlinuxfish, 2023-06-07:
I was looking at the documentation to learn how to flash the new firmware and noticed in the `to get started` section it references going to section 9.2 if you have a windows machine, but the headings below do not have section numbers.

It looks like a pandoc reference configuration issue if I were to guess, but I don't know where to find the wiki source.

--- 
### [Excerpt](https://github.com/ploopyco/headphones/wiki/Appendix-C%3A-Firmware-programming)
To get started:

- If you have a Windows machine:
	- Go to section 9.2.
    - Follow the steps until you complete section 9.2.4.
    - Read, but do not complete, section 9.2.5.

13tales, 2023-06-07:
I believe those refer to the sections in the Pi Pico pdf documentation, not the Ploopy Headphone docs. Unless I'm misunderstanding you?

mrlinuxfish, 2023-06-07:
Oh, you're right. I didn't notice it was pointing to another piece of documentation

closed 2023-06-07 by mrlinuxfish