ploopyco/headphones #20: DAC loses connection every ~15minutes

ploopyco, 2023-07-28:
We have a user report of the DAC disconnecting from the PC. 

Some notes:
- Windows PC, USB A to C cable shipped with kit used.
- USB port is on a desktop, attached to the front port panel.
- Disconnects happen on average every 15 minutes.
- Issue can be remedied reliably by unplugging the DAC and plugging it back in.
- No notable errors on the Windows side (pop-ups or whatnot).
- Seems to work better with a C to C USB cable, but testing is still in progress on that front.

randomlongname, 2023-10-26:
I am having the same problem with the headphones I just finished assembling yesterday (10/25/23). 
I have found the more I play the more frequent it resets, to the point it resets immediately. 
It does not reset with only one head phone attached.
Most importantly if I turn the volume down one notch from max the problem goes away completely. :)

qoh, 2023-10-30:
I find that this is strongly correlated with volume. I can reliably reproduce it by setting my OS volume level above 87% (-3.63 dB) and playing [Cleaning Apartment - Clint Mansell on Spotify](https://open.spotify.com/track/3nHWJEJlL1wUvvsQG7UTat), and waiting for the 00:04 low frequency "thunk" hit. That is, if I do this with a volume level of 88% (-3.33) dB, then the DAC disconnects at 00:04.

- Firmware PM16, default configuration
- Spotify volume = 100%; streaming quality = Very high; auto adjust quality = Off
- PipeWire PulseAudio on Linux

Future testing suggestion: Try it with a public raw audio file.

Other anecdata:
- I've also encountered this while raising Headphones Toolbox EQ curves higher
- It also occurs for me with a Pixel 6 Pro running Android 14

It's almost like an overflowing sample value is causing NaNs/infinities that brick the chain until a reset.

qoh, 2023-11-02:
A few more observations:

It is not on the input side: Lowering Pre-EQ Gain from -4.1 dB to -8 dB lets me crank OS volume to 100% without reset.

It is not in the filtering: Raising Post-EQ Gain from +3 dB to +6 dB causes it to reset there even at 87%.

In other words, in my experience, the DAC disconnects when a sample is too loud somewhere after the Post-EQ Gain step; its value does not matter anywhere before that.

ploopyco, 2024-04-07:
The root cause of this problem appears to be too much power draw resulting in the upstream USB host cutting power to the device. Reducing power draw by lowering the volume is a possible fix (though not ideal). Finding a USB power with higher power capacity (or no overcurrent protections) is another. Closing for now; if you're still having problems, please email us at contact@ploopy.co

closed 2024-04-07 by ploopyco