ploopyco/headphones #28: Add headphone jack audio input option

Renha, 2023-10-07:
While it is understandable to want to use single interface for both audio and charging, this limits the use cases of the headphones. For example, for delayless audio monitoring during the recording my microphone requires audio jack capable headphones.

ploopyco, 2023-10-16:
The DAC we use (PCM3060) has an ADC for this purpose, but the board as designed lacks an input connector and suitable pre-amp. As much as we wanted to, every design is an exercise in compromise, and we figured more of our users would want these headphones for listening than for live audio applications.

That said, the project is open source, and I'd be delighted to see a modified version of the PCB that has the requisite hardware to enable it. Closing in anticipation of a pull request. ❤️‍🔥

closed 2023-10-16 by ploopyco

RustoMCSpit, 2023-10-29:
> That said, the project is open source, and I'd be delighted to see a modified version of the PCB that has the requisite hardware to enable it. Closing in anticipation of a pull request. ❤️‍🔥

could this be mentioned on the website that the project is looking for someone to PR this? even in the readme would be great.