ploopyco/headphones #31: GitHub Organization?

PhoneDroid, 2023-12-06:
I was wondering if there was a reason you haven't converted  
the `ploopyco` GitHub account to a GitHub organization yet?

( Settings > Organization > Transform account )

I would also recommend enabling discussions on the organization, so  
there is a central place to discuss general ploopy topics on GitHub.

PhoneDroid, 2023-12-06:
Also, also, since there is not discussions tab yet ( GitHub Organization ),
I will just add it here:

Will there be more colors in the future ( for the printed Headphone parts ) ?

ploopyco, 2024-04-07:
Thanks for catching the typo. :) 

To keep the number of communications channels manageable, we're on Reddit (r/ploopy) and Discord (https://discord.gg/eYXvtpWBpn). Product support goes through email at contact@ploopy.co.

closed 2024-04-07 by ploopyco