ploopyco/headphones #37: Install instructions pedagogy

schneems, 2024-06-15:
## Current instruction pedagogy

The format of the install instructions is currently

- Header with a command i.e. "Apply a strip of tape to one of the Driver Flex Boards"
- Picture
- Bullet point of very important caveats that might be below the fold of the picture

## Suggested instruction pedagogy

I am suggesting a change to the format and order of instructions

- Header numbered or labeled without a verb or command i.e. "Heandband Step 4" or "Step 4" or "Section 2, Step 4" or "Step 2.4"

Why? The current instructions require me to read a command and then exercise executive function to not act on it immediately. It sounds minor, but if there's an order we want the user to follow i.e. "read all instructions, then follow command" then the user will have more success if the instructions are given in that order. Removing the "command" from the step name is progress towards that goal.

- Bullet point of very important caveats without any verbs

Why? This structure presents the user with what we want them to do before they take an action. I would also possibly suggest sub-dividing "really important" caveats into "verification" actions instead of being caveats.

This also moves the picture below the caveats because today it's common to have the caveats hidden by accident when following directions, for example: 

[image]

- Picture

Why? Move picture so it doesn't hide the caveats

- Command with explicit marker such as an emoji like ➡️ or since the ploopy logo looks like a pizza it could be 🍕

Why? Make commands as explicit as possible. Use consistency and visual markers to differentiate commands from other writing. I'm on the fence for if I think this should be before or after the picture i.e. grouped with the other writing or by itself. The upside of putting it after the picture is it's more likely to be identified as a command. The downside is someone might scroll so the picture and command are only visible and make it harder to reference the caveats.

## Other miscellaneous suggestions

The above doesn't really require any rewording, just re-ordering so I think it's not a huge change. One other suggestion is pulled from some advice I wrote a long time ago https://www.schneems.com/post/60359275700/prepare-do-test-make-your-technical-writing-shine which is to include "test" instructions.

I had an awful time with the magnets, that section has a lot of disclaimers

> (!) [WATCH THIS VIDEO on how to insert the magnets into the Driver Cap.](https://imgur.com/a/SkTC3ym)
> The magnets must be inserted into the Driver Cap in a very specific orientation. If you do not follow the video exactly, the Driver will be extremely quiet.
> Note that the loose magnet is attracted to the red magnet, and is lined up directly on top of it.
> Don't let the magnet twist around when you insert it into the slot.
> The magnets must be inserted in sequential order.
> Repeat the actions in the video for all remaining magnets. [Here is a second video on the magnet insertion process, in case it's still unclear.](https://imgur.com/a/YOAVskD)
> Again, each magnet should fit tightly into the Driver Cap. If it doesn't fit snugly, try wrapping a single piece of tape around the magnet and insert it again.

It's hard for the user to keep multiple pieces of information in their head. These many caveats could be broken into some "verification" steps. I would start with the last one. Instead of 

> each magnet should fit tightly into the Driver Cap. If it doesn't fit snugly, try wrapping a single piece of tape around the magnet and insert it again.

You could say:

- 💡 If the magnets are not tight in the cap they might "shoot" out or otherwise be difficult to install
- 🍕 Check that the magnets fit snugly in the cap by taking one magnet and inserting it in the cap as shown in the picture
- ✅ If the magnet is loose or falls out, wrap a single piece of tape around the magnet. 
- 🔁 Check that the magnet is now snug. If not, add more tape 🔁 . Use this same process on any future magnets that are not snug.

It's more work and words but it does a few things. The 💡 is giving the background context of why this is important, then the check is turned into an action item in 🍕. Then the ✅  becomes a verification step. For anyone who's magnets were snug, great they can skip to the next section, but now everyone has for sure done a test fit and is aware that there's a mitigation available if a future magnet doesn't fit.

Anyway, I love the idea of open hardware and I spend a lot of time writing software tutorials. The existing tutorials are already good, I wanted to share my experience of how I think we could make them better. Thanks for putting all this together!

ploopyco, 2024-06-17:
This is actually great advice. We're reworking all of our documentation, and I'm going to be referring to this when I'm doing it. Thanks for putting this all together!

schneems, 2024-06-17:
Thanks again for the docs, and validating my feedback. You're welcome to close the issue and reference it (or keep it open if you want until the re-write is done).