ploopyco/headphones #39: [Feature Investigation] Different magnet alignment: Halbach Array

schneems, 2024-07-08:
Someone suggested a different magnet alignment for stronger magnetic field. I suggested they open an issue, but didn't so here I am. Context https://www.reddit.com/r/ploopy/comments/1dr42ay/halbach_array_for_headphones/.

The idea is that by changing how the magnets are aligned, it could make the headphones stronger. I didn't have a chance to try it out. I'm curious if it's a known bad idea because of **reasons** and if it's been tried or not.

Can someone try this out and report back? I'm swamped at the moment, but I'm interested in learning more.

cab404, 2024-08-13:
I've slapped together a model allowing you to put 7 magnets in for a proper Halbach Array.
https://github.com/cab404/ploopy-mods/blob/master/newcap/experiments/hallbachcap.stl

[image]
[image]

Should be printable.
IDK, maybe print it and compare to baseline?

cab404, 2024-08-20:
[image]
something-something you will require 13x 3x3x30 brick/cylindrical magnets

schneems, 2024-11-18:
I got a bit distracted in my quest for louder jams.

I found https://github.com/cab404/ploopy-mods/tree/736a983d29a48fdbb119229f41d20266aa45e459/newcap from the Subreddit some time ago but haven't gotten around to trying any of them. Seems that it "simply" moves the magnets closer to the driver which increases the volume. It requires modifying the headphones though:

> You'll need to remove several perimeters (0.5mm) from the sides of driver frames, so the cap can with with magnets.

(And just noticed that @cab404 made that! Thanks! 👋 )